SJSURoboticsTeam/urc-control-systems-2024 #34: LCD display

Viha123, 2025-08-22:
- this is currently low priority but it needs to be tested. 
- @MichaelYKersey could you please update here what body of work needs to be done, so we can assign this to someone new.

MichaelYKersey, 2025-08-24:
The LCD display code is [here](https://github.com/MichaelYKersey/urc-control-systems-2024/tree/nhd0420d3z) who every get assigned can make a new branch and run `git pull https://github.com/MichaelYKersey/urc-control-systems-2024.git nhd0420d3z'. The code should be mostly done, it just needs to be tested ,debugged, and update the access modifier for some of the methods (some stuff in the demo is commented out and some methods were made public form debugging). The most notable thing to note when testing is that the LCD Display needs a 5V i2c not 3.3V (it's why work on the LCD display driver got held up) so you need to check that the i2c you are using is 5V (I think the hub or drive board is the board that has the 5V i2c). It might be as simple as reverting to a previous commit (with out debugging modifications) and testing it, or you might have to do some debugging and reading the datasheet to figure out where I messed up.

Viha123, 2025-09-14:
@rdezsol we may need like a 5V level shifter for this.

LQR471814, 2026-10-17:
Tested again in #88

closed 2026-10-22 by LQR471814